ratgdo/esphome-ratgdo #394: there is no driver.....

artlessknave, 2025-04-01:
[/](https://ratgdo.github.io/esphome-ratgdo/)

page claims that the drivers are included and doesnt provide any drivers, but there is no driver. so the device is a paperweight...

[image]

[image]

[image]

going to try linux mint, but thats pretty annoying that there IS no driver

PaulWieland, 2025-04-01:
This isn't a firmware issue. Please reply to your receipt to open a support ticket.

closed 2025-04-01 by PaulWieland

artlessknave, 2025-04-03:
OK. i finally found the driver on this page:

https://esphome.io/guides/physical_device_connection.html

which isnt mentioned AT ALL on this page:

https://ratgdo.github.io/esphome-ratgdo/

specifically, i needed the cp210x which is here:

https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

also, I couldnt find the support email until i found the keyword ratcloud, as the email didnt show up at all when i searched for ratgdo for some reason.
attempting the firmware. putting this here in the hopes that someone else having this issue might find it.