ratgdo/esphome-ratgdo #399: Download OTA Firmware links leads to 404

damounh, 2025-04-02:
When navigating to download the OTA firmware from https://ratcloud.llc/pages/firmware#upload_firmware, the download link goes to a 404: https://ratgdo.github.io/esphome-ratgdo/v25iboard-esp8266.ota.bin?

jprowland, 2025-04-02:
Same problem.

PaulWieland, 2025-04-02:
Its not just the download firmware link. It's also the firmware installation itself. 

[image]

@bdraco It seems to have been broken with the last merge proposed by the dependabot related to `github-actions/upload-artifact`. Still looking for the exact reason though

PaulWieland, 2025-04-02:
I suspect the issue is in the `Build Firmware` step under `Copy firmware files to folder`
[image]

The previous deployed build looked like this:

[image]

PaulWieland, 2025-04-02:
For an unknown reason the behavior of `cp -R firmwares/*/* output/` changed in the build environment. Before it would put all matched files directly into the output directory (confirmed that's how macOS works as well). Now it preserves the directories when copying.

[Added a workaround](https://github.com/ratgdo/esphome-ratgdo/blob/5793e16c14357e0e12fd523abd46ef3eba4f318b/.github/workflows/build.yml#L116C11-L116C13) `mv output/*/*.bin output/` to remove the sub directory.

damounh, 2025-04-02:
awesome thank you @PaulWieland ! binaries accessible now. please feel free to close the issue, otherwise I can close it later tonight.

closed 2025-04-02 by PaulWieland