ratgdo/esphome-ratgdo #403: ratgdo32 not working on supported door openers

jimfrety, 2025-04-09:
I have just purchased 3 ratgdo32's.
I’m having trouble with these items not working on my yellow button Chamberlain roller door openers. I have installed the Security + 2.0/ratgdo32 firmware.
The previous devices i purchased from you (ratgdo v2.53i kit) work on these same doors, but I was unable to order those as you seem to have discontinued them.

I have wired them the same way and ensured the doors are working correctly etc and as I said, fitting the v2.53 ratgdo device from my other door works of the new doors.
Please see attached model details of the door opener and the wiring configuration.  Please note the Chamberlain website says these doors are Security + 2.0 https://www.chamberlaindiy.co.nz/product/rollerlift-plus-cr855myq-2/#specs-tabs3

I have tried using the devices via the web interface and via HA - neither work.
Any help would be appreciated.

Thanks
Jim

PaulWieland, 2025-04-09:
As far as I'm aware, all roller door openers use the Dry Contact control protocol. Please switch to that firmware and try the toggle button in the ESP Home firmware to verify that the door responds.

jimfrety, 2025-04-09:
that fixed it - thanks

closed 2025-04-09 by jimfrety