ratgdo/esphome-ratgdo #404: [secplus1]: receive timeout seems to be too short

troian, 2025-04-25:
My garage door openers are on Security+ and were logging `Discard incomplete packet: [38 ...]`.
Everything like lights, door toggle works, however, it was never able to sync with door status.

Out of curiosity, I just set timeout to 1s (I get it is way to big, just for debug reasons) and incomplete packet message no longer logs as well as door status syncs perfectly.

I guess `Secplus1::read_command()` might need some tweaking, also, would likely be good option to have read timeout as config parameter

PaulWieland, 2025-04-25:
It sounds like for some reason your door control panel's polling frequency, or the GDO itself is slower than usual. Try disconnecting the wall control and let ratgdo go into emulation mode (it has to be rebooted after you disconnect the wall control). See if the door status works in emulation mode with the 100ms timeout. That will indicate if its the door opener or the wall control.

troian, 2025-04-25:
Thanks, will check. I'm waiting for a replacement panel as well.

Not quite related question. It looks like software UART is being used, any reasons behind it? esp32dev does have 2 hardware UARTs?

closed 2025-04-27 by troian