ratgdo/esphome-ratgdo #409: ESPHome-Ratgdo no longer working. Buttons to raise/lower and all other entities grayed out.

rdperkins, 2025-05-01:
Like it says. It hasn't worked for the past couple of weeks.

PaulWieland, 2025-05-01:
The entities in the ratgdo web UI cannot be grayed out, so I am assuming you are referring to another platform such as home assistant. 

What happens when you visit the ratgdo web UI directly?

rdperkins, 2025-05-01:
Yes, the home assistant integration where it shows the controls. When I try to login to the device ip, it won’t connect, and I checked the ip address in my Dream Router and it’s active.

***@***.***
Robert Perkins
10486 Bethpage Drive
Indian Land, SC 29707
(704) 907-9480
***@***.***

rdperkins, 2025-05-03:
I don't see the ratgdo device in my router client list. how can i find it's ip or dn?

rdperkins, 2025-05-03:
[image]

breakingspell, 2025-05-07:
Hello! Chances are the device is simply offline, so Home Assistant is marking it as unavailable. Sounds like it just needs a fresh flash. 

On the ratgdo:
- Flash the newest firmware via USB: https://ratgdo.github.io/esphome-ratgdo/. 
- Access the device via SSID and configure your network with the captive portal wizard. 
- You should be able to see the device in your router's DHCP/client list and access directly via IP (no login).

In Home Assistant:
- Navigate to `Integrations` (not `Entities`) and select ESPHome. 
- The integration should automatically see the ratgdo on your network. 
- You may have to remove the old device, but it may update the existing entry for you.

rdperkins, 2025-05-14:
> It sounds like it's just offline or the network is preventing it from connecting, in which case it should go into captive portal/hotspot mode.
>
> If you do need to re-install, the firmware installation instructions are here: https://ratcloud.llc/pages/firmware
> If using USB, you can configure the wifi directly from the web installer.

I have the 2.531 version which I reflashed. During install I gave it my ssid and pw and things seemed to go ok except the redirect to HA esphome bombed with a 404. I tried to adopt from HA ESPHome and got this
INFO Successfully compiled program. INFO Resolving IP address of ratgdo-garage-door-opener.local in mDNS INFO Resolving IP address of ratgdo-garage-door-opener.local ERROR Error resolving IP address of ratgdo-garage-door-opener.local. Is it connected to WiFi? ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips) ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

Robert Perkins
10486 Bethpage Drive
Indian Land, SC 29707
(704) 907-9480
***@***.***

rdperkins, 2025-05-25:
Happy Memorial Day. Sorry to intrude. I've followed all the instructions staring with re-flashing, setting up Wi-Fi, checking to see if I have a working ip on my UDR router, device and entities recognized by ESPHome in HA. Then I powered down the gdo, hooked up the ratgdo, powered it up and rechecked the ip. When I first power it up, the gd opens partially then closes. The up/stop/down entity doesn't work and the up arrow is grayed out. The only wiring change I made was to place the white (yellow on the picture below) into the right of the two white ports. Does this matter? I've spent at least a dozen hours trying to get this to work again over the past 3-4 months. Very frustrating. Thanks for your help.

Here is my gdo connection photo:

[IMG_2010.jpg]

Here are snapshots from HA
***@***.******@***.***

Robert Perkins