ratgdo/esphome-ratgdo #410: Gate_Party_Mode - YAML Troubleshooting

TheRandy, 2025-05-02:
So I feel like I may be the only one looking to implement the party-mode option that is discussed on the RATGDO website for gate operators.  At the very bottom of the 'wiring' section...it discusses the gate operator installation AND leads to this repository for the 'custom' .yaml to get party-mode working.

Only thing is...the yaml points to i think the 2.5i board and based on my testing, clearly has different pins as the 'custom' yaml does not work for the ratgdo32 board.

Any chance someone else has done this and can assist?

PaulWieland, 2025-05-03:
[Added an example for ratgdo32](https://github.com/ratgdo/esphome-ratgdo/tree/main/static/customized_examples). I haven't personally tested it so please let me know the outcome.

TheRandy, 2025-05-04:
@PaulWieland - that worked beautifully!  You're awesome!  Thanks. Now I'm fully functional from within Home Assistant and HomeKit with a switch for 'Party Mode'.  Tested and worked flawlessly.

Thanks again.

closed 2025-05-05 by PaulWieland