ratgdo/esphome-ratgdo #412: Want to use dry-contact inputs as sensors and NOT trigger actions on a Sec+ 1.0 chamberlan. (esphome)

PedanticAvenger, 2025-05-05:
So I'm trying to solve a potential physical security challenge.  I haven't encountered this myself yet but my neighbor had someone manage to pull the emergency release on his door and got into his garage.  This got me to thinking that this is a scenario that ratgdo can't alarm on as the gdo didn't do anything to open the door.
I have a magnetic sensor on my door that isn't wired into anything and I'd like to use it into the dry contact open input and NOT have it immediately open the door as soon as it closes. :)

I suspect I can just go mess with the yaml and end up with what I want but don't want to strand myself away from mainline code/config updates if I can avoid it.  Have I missed something and this is easy or suggestions?

PaulWieland, 2025-05-05:
You could connect the switch to the TL input and base your security notification off of that input being triggered.

PaulWieland, 2025-05-13:
@breakingspell I really don't understand what you mean. The use case here is for Security + 1 and 2 openers where the door can be physically opened by pulling the release cable and opening the door. When this occurs, the GDO itself doesn't know where the door physically is which is why it would be useful to have a dry contact security switch to create an alert when the GDO reports the door state is closed but the switch is opened.

This is irrelevant for dry contact controlled doors (assuming you set the limit switches up correctly). As soon as the closed limit switch opens, the door state becomes opening and remains that way until the open limit switch is closed (door state = open) or the close limit switch is closed again (door was reversed mid travel and the door state = closed).

breakingspell, 2025-05-13:
> The use case here is for Security + 1 and 2 openers where the door can be physically opened by pulling the release cable and opening the door. When this occurs, the GDO itself doesn't know where the [...]

This is also the case for my dry-contact opener (Sommer/Synoris 550), the trolley can be manually disconnected from the track with a pull string. A dedicated "security switch" would be useful regardless of the protocol. 

> As soon as the closed limit switch opens, the door state becomes opening and remains that way until the open limit switch is closed (door state = open) or the close limit switch is closed again (door was reversed mid travel and the door state = closed).

Yes, and that's actually something I want to correct for in the dry contact firmware. If the closed limit switch opens without a ratgdo call having invoked it (wall button or cord pull), use a timeout based on the set duration and act accordingly if either limit switch hasn't been reached in that duration.

PedanticAvenger, 2025-05-15:
This interesting discussion aside, that did seem to work for me (with odd light behavior on gdo until an open/close cycle.
But after a couple days of that working it seems my unit has died or at least tried to.  I'll have to pull it down and check the esp.