ratgdo/esphome-ratgdo #414: Update code for changes coming in ESPHome

bdraco, 2025-05-12:
https://github.com/esphome/esphome/pull/8747

RyanMorash, 2025-05-21:
Here is the developer blog post: https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/

I'm getting warnings for the lock schema and cover schema.

bdraco, 2025-05-21:
We probably need to wait a bit to do this as we need to set a minimum ESPHome version when we do as well

PaulWieland, 2025-06-25:
Added: https://github.com/ratgdo/esphome-ratgdo/tree/schema-update
Doesn't build yet.

gcs8, 2025-06-28:
Yep, also here just to add to the notes I was seeing.

```
INFO ESPHome 2025.6.2
INFO Reading configuration /config/esphome/ratgdov25i-e6863a.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo.git@main
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
WARNING Using `lock.LOCK_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `lock.lock_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: ratgdo
WARNING Using `cover.COVER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `cover.cover_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: ratgdo
INFO Detected timezone 'America/New_York'
```

PR #439 (2025-07-14): fix: update deprecated ESPHome schemas for 2025.6.0 compatibility
## Summary

This PR updates the esphome-ratgdo component to use the new ESPHome schema functions, replacing deprecated uppercase schema constants that will be removed in ESPHome 2025.11.0.

## Changes

### Schema Updates
- Updated `lock.LOCK_SCHEMA` → `lock.lock_schema()` in `components/ratgdo/lock/__init__.py`
- Updated `cover.COVER_SCHEMA` → `cover.cover_schema()` in `components/ratgdo/cover/__init__.py`

### Configuration Updates
- Set minimum ESPHome version to `2025.6.0` in all base YAML files:
  - `base.yaml`
  - `base_dry [... cut]
PR #445 (2025-07-15): fix: update deprecated ESPHome schemas for 2025.6.0 compatibility (attempt 2)
Fixes #414
closed 2025-07-15 by bdraco