ratgdo/esphome-ratgdo #416: Feature Request:  Disable Dry Contacts

tangopoint, 2025-05-17:
I would like ratgdo to have the option to disable dry contacts.   My ratgdo regularly receives phantom dry contact triggers and opens the garage door.   The installation otherwise is working fine, but is not usable due to the random openings.   The unit is correctly connected and the keypad was changed for troubleshooting.   I am using a ratgdo 2.5 with a Liftmaster 8500W Security 2.0.   I am not the only one to have this issue.

If the user had the option to disable dry contacts if they are not being used, it would be a more robust and secure installation.

PaulWieland, 2025-05-17:
What does the ratgdo log show when the door phantom opens?

tangopoint, 2025-05-18:
Here are the entries from the ratgdo event log:

3272250	ping		
12:35:29.675
3274089	log	[0;36m[D][binary_sensor:036]: 'Dry contact open': Sending state ON[0m	
12:35:31.519
3274089	state	{"id":"binary_sensor-dry_contact_open","value":true,"state":"ON"}	
12:35:31.577
3274192	log	[0;36m[D][cover:076]: 'Door' - Setting[0m	
12:35:31.642
3274193	log	[0;36m[D][cover:084]: Position: 100%[0m	
12:35:31.643
3274395	log	[0;36m[D][number:012]: 'Rolling code counter': Sending state 22101.000000[0m	
12:35:31.827
3274395	state	{"id":"number-rolling_code_counter","value":"22101","state":"22101 codes"}	
12:35:31.889
3275019	log	[0;36m[D][binary_sensor:036]: 'Dry contact open': Sending state OFF[0m	
12:35:32.459
3275019	state	{"id":"binary_sensor-dry_contact_open","value":false,"state":"OFF"}	
12:35:32.516
3275705	log	[0;36m[D][ratgdo:113]: Door state=OPENING[0m	
12:35:33.136
3275708	log	[0;36m[D][ratgdo:222]: Light state=ON[0m	
12:35:33.192
3275708	log	[0;36m[D][ratgdo:228]: Lock state=UNLOCKED[0m	
12:35:33.193
3275711	log	[0;36m[D][ratgdo:235]: Obstruction: state=CLEAR[0m	
12:35:33.193
3275715	log	[0;36m[D][ratgdo:207]: Learn state=INACTIVE[0m	
12:35:33.194
3275715	state	{"id":"light-light","state":"ON","color_mode":"onoff","color":{}}	
12:35:33.195
3275728	log	[0;36m[D][cover:170]: 'Door' - Publishing:[0m	
12:35:33.196
3275729	log	[0;36m[D][cover:173]: Position: 0%[0m	
12:35:33.196
3275729	log	[0;36m[D][cover:186]: Current Operation: OPENING[0m

PaulWieland, 2025-08-14:
You should be able to update your yaml and use the [`!remove` syntax](https://github.com/esphome/esphome/pull/5821) to remove the TO/TC inputs.

```
binary_sensor:
  - id: !remove ${id_prefix}_dry_contact_open
  - id: !remove ${id_prefix}_dry_contact_close
```

tangopoint, 2025-08-14:
> Also interested in this feature. I'm using it with Chamberlain Security 2+
> overhead garage door opener. Once every few weeks, the "Dry Contact Open"
> would be turned on and the garage door opens itself.

I replaced my ratgdo and it resolved the problem.   At one point the
ratgdo was improperly wired which likely damaged it and was causing the
issue.

closed 2025-08-14 by PaulWieland