ratgdo/esphome-ratgdo #417: Chamberlain Whisper Drive HD600 Wiring.

dkhill, 2025-05-20:
I am attempting to use the ratgdo32disco controller on my Chamberlain Whisper Drive HD600.  I followed the wiring diagram to a "T".  I was able to gain access to it and pair it with my Wi-Fi and I am getting a status page.  (Nice page by the way).
I can control the garage light, and garage door from the wall switch with the ratgdo32 in circuit.  But for some odd reason it just doesn't trigger the garage door to open or close.  I have two obstruction sensors (that I pointed two each other due to sunlight issues).  Sensor circuit uses 11.63 Volts, which should be well in the range.  I've wired the garage door wall switch through the 4 post terminal as per the diagram and have the Red / White wire going directly into the GDO.

Attached are some images and documents.

I'm so excited to get this thing to work!!

:)

-david hill

[image]
[image]
[image]

[Chamberlain - Garage Door Opener - HD600 - 114A4258.pdf](https://github.com/user-attachments/files/20320442/Chamberlain.-.Garage.Door.Opener.-.HD600.-.114A4258.pdf)

PaulWieland, 2025-05-22:
What does the log display?

dkhill, 2025-05-29:
Tried with dry contacts logs below.
18:17:26 | [D] | [cover:084] | Position: 100%
18:17:26 | [W] | [ratgdo_dry_contact:094] | The door is not closed. Ignoring door action: OPEN
18:17:35 | [D] | [cover:076] | 'Door' - Setting
18:17:35 | [D] | [cover:084] | Position: 0%
18:17:35 | [W] | [ratgdo_dry_contact:098] | The door is not open. Ignoring door action: CLOSE
18:17:57 | [D] | [switch:012] | 'LASER' Turning ON.
18:17:57 | [D] | [switch:055] | 'LASER': Sending state ON
18:18:03 | [D] | [switch:016] | 'LASER' Turning OFF.
18:18:04 | [D] | [switch:055] | 'LASER': Sending state OFF
18:18:07 | [D] | [switch:012] | 'LASER' Turning ON.
18:18:07 | [D] | [switch:055] | 'LASER': Sending state ON
18:18:10 | [D] | [switch:012] | 'LED' Turning ON.
18:18:10 | [D] | [switch:055] | 'LED': Sending state ON
18:19:04 | [D] | [ratgdo:113] | Door state=OPEN
18:19:04 | [D] | [cover:170] | 'Door' - Publishing:
18:19:04 | [D] | [cover:173] | Position: 100%
18:19:04 | [D] | [cover:186] | Current Operation: IDLE