ratgdo/esphome-ratgdo #423: Ratgdo 2.5 ESPHome update fails in HAOS for all 3 Ratgdo devices

NukeThemTillTheyGlow, 2025-06-03:
I have 3 Ratgdo v2.5 that have been working flawlessly at my house for the past year plus.  In the past few days I've had updates take place on HAOS as per normal and then an ESPHome Builder update that per usual triggers 3 separate

[logs_update-all.txt](https://github.com/user-attachments/files/20581712/logs_update-all.txt)

 ratgdo device updates which I normally do all at once via the "update all" command within the ESPHome Builder screen.  Below is the log for one of the 3, the messages are idential.  I also am not able to open the webpage for any of the 3 devices from the ESPHome Builder device listing either.

Now I am also having Unifi network issues after what appears to be a buggy Protect update that caused my Network and Talk apps on my UDM Pro to stop working.  Remains unresolved as of 7 pm ET today, but I was able to get the Network app up and running again after applying latest update and it appears Network is functioning normally.  This started yesterday about 7 am when I tried to do the Protect update and then later yesterday I did the HAOS ESPHome Builder update normally (worked), but when I went to ESPHome Builder to do the update all, it came back with 3 failures, first time I've ever seen that.  I thought it might be related to not being able to resolve ip address's or something, but apparently not.

Full log file attached.

Installation method
Home Assistant OS
Core
2025.5.3
Supervisor
2025.05.3
Operating System
15.2
Frontend
20250516.0

Updating /config/esphome/ratgdov25i-e409a7.yaml
------------------------------------------------------------

INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/ratgdov25i-e409a7.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
WARNING Using `lock.LOCK_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `lock.lock_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: ratgdo
WARNING Using `cover.COVER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `cover.cover_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: ratgdo
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing ratgdov25i-e409a7 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Secplus @ 1.0.0+sha.f98c322
|-- EspSoftwareSerial @ 8.1.0+sha.57f1aa7
|-- noise-c @ 0.1.6
|-- Improv @ 1.2.4
RAM:   [=====     ]  47.4% (used 38840 bytes from 81920 bytes)
Flash: [=======   ]  65.3% (used 681517 bytes from 1044464 bytes)
========================= [SUCCESS] Took 1.75 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ratgdov25i-e409a7.local in mDNS
INFO Resolving IP address of ratgdov25i-e409a7.local
ERROR Error resolving IP address of ratgdov25i-e409a7.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
====== [ERROR] /config/esphome/ratgdov25i-e409a7.yaml ======

PaulWieland, 2025-06-03:
`ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond `

MDNS didn't resolve. It can't find the ratgdo host. Can you ping it from your computer?

NukeThemTillTheyGlow, 2025-06-03:
Yes, I just pinged each one from my main home pc.  I'm trying to think what might have changed, either in HAOS or Unifi.  the network assignments, vlan's, etc in unifi have been in place forever, no changes there other than the issues with the Protect app update and associated issues. 

Let me try to ping them from HAOS

NukeThemTillTheyGlow, 2025-06-03:
> Yes, I just pinged each one from my main home pc. I'm trying to think what might have changed, either in HAOS or Unifi. the network assignments, vlan's, etc in unifi have been in place forever, no [...]

Successfully pinged all 3 from the HAOS terminal

NukeThemTillTheyGlow, 2025-06-03:
I bet you its the most recent UniFi OS update, 4.2.12, changed something.  Looking at the release announcement, it specifically mentioned mDNS.

[](https://community.ui.com/releases/UniFi-OS-Cloud-Gateways-4-2-12/7603848f-1dc4-4228-9cdd-459f9f8b017a)

NukeThemTillTheyGlow, 2025-06-03:
> I bet you its the most recent UniFi OS update, 4.2.12, changed something. Looking at the release announcement, it specifically mentioned mDNS.
> 
> [https://community.ui.com/releases/UniFi-OS-Cloud-Gateways-4-2-12/7603848f-1dc4-4228-9cdd-459f9f8b017a](url)

I have not changed my VLAN and other related settings in a number of years.. the Ratgdo's all connect via wifi to my IoT network.  I know that Unifi just came out with "Traffic Flows" and my guess is something changed.  I need to grab some dinner, but my guess is its in here somewhere.  Updated was 5/22 so it has not been that long, I've not done an ESPHome update since that update.  and by the way, all 3 ratgdo's respond to my HAOS dashboard commends

NukeThemTillTheyGlow, 2025-06-04:
I just went through all 10 pages of the UniFi OS release notes.  While I didn't find exactly the same problem, I found enough "close enough" issues that its clearly the UniFi OS update from 5/23.  I hesitiate to start changing things right now with no clear diagnosis, but after I get a few other things done, I might take a few cracks at it.  If anyone has has any suggestions, I'm all ears.  I did confirm that my IoT network has both ICMP snooping turned on as well as Multicast DNS.  I restarted the wifi AP in my garage.

NukeThemTillTheyGlow, 2025-06-04:
Something changed in the past 10 days or so with my network and it has to be UniFi software/firmware update related.  I tried adding statis_ip and related variable to one of my ratgedo yaml definitions and best I can do is now get an error opening a port number message.    Love to hear & see working configs in UniFi nextworks right now.

jdranchman, 2025-06-04:
I current on my unifi bits (official) and have my IoT on a different IP
space but do not yet have access rules to block stuff off. My ratgdo disco
is running as expected and took an upgrade yesterday with the current
esphome builder. I haven't done a full test but I can get to the web
interface (reserved that IP some time ago) via mDNS.
I'll give it a full few cycles and whatnot Thursday.

Jim Davis

PaulWieland, 2025-06-23:
There is an application called “Discovery” that was formerly called Rendezvous Browser. It shows all discovered mdns entries. I think it’s only available for iOS and macOS though.

https://www.tildesoft.com/#RendezvousBrowser

NukeThemTillTheyGlow, 2025-06-25:
Resolved.   I just spent a couple of hours on a TeamViewer session with UniFi support.  My original gut feel was correct, the issues I originally had were caused by the updates related to the Unifi OS Console update to my UDM-Pro AND the Network app update including the new Zone Based Firewall.  Here's what happened:

- The issues with the Network app not starting due to being out of disc space becuase of a bug in the backup service not cleaning up and deleting old backups was the original issue.  This caused the Protect and Talk apps to not start either.  But basic networking and camara display worked, you just could not change anything or manage it all.  Fixes applied by Unifi support after they SSH'd into my console.
- But then my 3 ratgdo openers showed as offline in ESPHome Builder and not functional from HA dashboard.  They were originall config on my IoT network/VLAN using DHCP and that had not changed in probably 18 months.   So after tones of posts, message and research, I moved them back to my core LAN VLAN and on advice of others, changed to static IP.  Reflashed, updated in HA and all good.
- Finally Unifi support said with all of the other issues behind us, lets see if we can fix your "mDNS issues".   No mDNS issues!  Setup properly.  I tried to find and use a number of Windows based Bonjour or MDNS analysis tools, but nothing would work.  I was going to try Wireshark, but ran out of time and decided to just talk to Unifi support. 
- Unifi support rep connected via Teamviewer, we reviewed all of my settings and all of my basic mDNS, VLAN, firewall rules were fine.  We disabled my firewall rules, I moved one of the ratgdo's back to the IOT VLAN using static ip and it worked fine!  
- Then support rep started to turn on all firewall rules.  On the last one, it wouldn't start.  He called it redundent with other rules, but given that it used to start and work in the past, the only change is the addition/changes related to implementing Zone Based Firewall.
- His advice was: Move your remaining Ratgdo's back to IOT and they work great again!   But read up on the new Zone Based Firewall system and how to implement the basic capability that I had in place with the policy based/original rules.  Then delete all of the policy based rules, enable ZBF and implement the same thing via ZBF.  Once that is working, I'm going to create new vlan's for my work and maybe a Unifi Protect cameras.

NukeThemTillTheyGlow, 2025-06-25:
> Update: Last week I took down each each ratgdo, reflashed it and went with both static IP addresses and moved them from my IoT network to my core LAN network. I have not changed or updated my firewall rules in a number of years when I did a fresh install on my UDM-Pro after moving from a cloud key. After some trial & error, all are working fine except I noticed only one update would show up when Esphome updates came out, but I just googled that and enabled the firmware so I'm sure thats going to be fine going forward. And after checking, my Sonos devices were never moved from core LAN to IoT....so at this point, I'm not really sure what devices on the IoT VLAN are actually using mDNS.
> 
> **Any good way to test mDNS traffic across VLAN's other than ping the device?**
> 
> Ubiquity support resolved the UniFi OS Console & Network & Protect app issues and then created a new support ticket for me to resolve the mDNS issues. So my device config has changed, so its not a 100% apples to apples comparison. Here is what I plan to do to test, I really don't want to create a big project, but would like to get classic IoT devices on the IoT VLAN and working...has me wondering if I need to reevaluate all of the firewall rules too. I'm going to set up the Teamview session with UniFi support and make sure mDNS is working and then move the ratgdo's back to the IoT VLAN.....
> 
> **is there a way to change the IP address of them without having to take down, flash, reinstall?**

Yes, this was easy.  1) change the YAML settings for each device in ESPHOme Builder.  I just changed the ssid, password and changed static ip detail to point to my IOT network.  2) save and install.  it will reboot using new IP 3) go to the Esphome device and reconfigure..change the ip address to the new one. 4) complete all changed and restart HAOS.

closed 2025-06-25 by NukeThemTillTheyGlow