ratgdo/esphome-ratgdo #429: Entity Not Found error 'lock.LOCK_SCHEMA

sgansky, 2025-06-25:
I have 3 systems installed and all were 100%.  At some point 1 system began showing an "Entity Not Found" error.  When  I update this door I get:
"Warning Using 'lock.LOCK_SCHEMA' is deprecated and will be removed in ESPHome 2025.11.0 Please use 'lock.lock_schema)...)' instead. 

Not sure where to correct this error.  I'm not a programmer. More of an electronics geek.  The other 2 doors are fine.

Thanks for any help

closed 2025-06-25 by PaulWieland

sgansky, 2025-06-25:
> Closed #429 <https://github.com/ratgdo/esphome-ratgdo/issues/429> as duplicate.

Paul

Can you help me with this.  Will an update just fix this down the road ?

Steve

Stephen Gansky
***@***.***
C) 215-990-5300

PaulWieland, 2025-06-25:
The warning you posted is just a warning. It doesn't affect anything at all functionality wise.