ratgdo/esphome-ratgdo #431: Compile error after esphome 2025.6.3 Home Assistant

SamuelCarson, 2025-07-04:
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)

PaulWieland, 2025-07-04:
I don’t think there’s an issue. [It’s building fine here](https://github.com/ratgdo/esphome-ratgdo/actions/runs/15963936244).  Clear your cache or create a new folder with the yaml in it and try again.

SamuelCarson, 2025-07-05:
Here is current yaml. Compiling under Home Assistant Esphome Builder. Do I have the correct github command? This worked until Esphome Builder 2505.6.3.

substitutions:
  name: ratgdov25i-87b47f
  friendly_name: ratgdov2.5i 87b47f
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: 3zNHZ3nkFBSTzTOopWxN1yaLiGxe+YPi/d7yWvtB+5o=

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Air-Gradient Fallback Hotspot"
    password: !secret fallback_ssid_password

captive_portal:

bdraco, 2025-07-05:
It's just a warning that you can ignore, it will eventually be solved in ESPHome