ratgdo/esphome-ratgdo #432: Compile Error: WiFi.h Not Found When Building with ESPHome (ESP32)

rotary65, 2025-07-06:
**Describe the bug:**  
When attempting to compile the Ratgdo integration with ESPHome on an ESP32 board, the build fails with a fatal error indicating that `WiFi.h` cannot be found. This occurs during the compilation of the `ESPmDNS` library.

**To Reproduce:**  
1. Use the following environment:
   - **ESPHome version:** 2025.6.3
   - **Board:** esp32dev
   - **Framework:** Arduino
   - **Platform:** platformio/espressif32@5.4.0
   - **Ratgdo integration:** gelidusresearch/esphome-ratgdo.git@1.0.0
2. Attempt to compile the configuration.

**Observed Output:**
```
Compiling .pioenvs/gd01/lib915/ESPmDNS/ESPmDNS.cpp.o
/data/cache/platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.cpp:42:10: fatal error: WiFi.h: No such file or directory
 #include "WiFi.h"
          ^~~~~~~~
compilation terminated.
*** [.pioenvs/gd01/lib915/ESPmDNS/ESPmDNS.cpp.o] Error 1
```

**Expected behavior:**  
The build should complete successfully, including all necessary dependencies for ESP32 WiFi functionality.

**Additional context:**  
- The error message suggests that the build system cannot locate the `WiFi.h` header, which is required by the ESPmDNS library for ESP32.
- The ESP32 Arduino framework should normally include `WiFi.h` by default.
- The PlatformIO output recommends searching for the library in the registry, which may indicate a misconfiguration or missing dependency in the build environment.

**Attachments:**  
- Full compile log
- ESPHome YAML file

**Possible causes:**  
- The Arduino framework for ESP32 may not be properly installed or configured in PlatformIO.
- There may be a missing or incompatible platform or library version.
- The Ratgdo integration or its dependencies may not be specifying the correct platform requirements.

**Steps attempted to resolve:**  
- Verified platform and framework versions.
- Checked that the board and environment match ESP32 requirements.

**Request:**  
Please advise on how to resolve the missing `WiFi.h` error or update the integration to ensure all dependencies are properly included for ESP32 builds.

**References:**  
[gd01.yaml.txt](https://github.com/user-attachments/files/21087440/gd01.yaml.txt)

[logs_gd01_compile.txt](https://github.com/user-attachments/files/21087439/logs_gd01_compile.txt)

bdraco, 2025-07-06:
Please report the problem to the fork gelidusresearch/esphome-ratgdo.git@1.0.0

You are not using this project

closed 2025-07-06 by bdraco

rotary65, 2025-07-06:
My apologies. I will follow up with the fork project.