ratgdo/esphome-ratgdo #450: Not sync after update

georgea62, 2025-07-16:
Board 2.5i, Security + 1.0 (red button, analog panel)
It was working for months with no issues.

The problem started today, I tried to update with the latest 2025.6.3, after this update it stopped communicating with the GDO.

The interest part in log is:
......
..........
[19:18:37][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[19:18:39][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[19:18:41][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[19:18:41][D][ratgdo_secplus1:077]: No wall panel detected. Switching to emulation mode.
WARNING ratgdov25i-47a1b3 @ 192.168.21.108: Connection error occurred: [Errno 104] Connection reset by peer

after this point it reboots and keep cycling with the same pattern.

The yaml is very basic:

substitutions:
  name: ratgdov25i-xxxxx
  friendly_name: ratgdov2.5i xxxxx
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxx
wifi:
  ssid: xxx
  password: xxx

I tried to clean-build and update many times.

What else I can try?

Thank you!

bdraco, 2025-07-16:
Can you post a stack trace?

https://esphome.io/guides/troubleshooting.html

bdraco, 2025-07-16:
Sadly it has to be done over a serial (USB) cable because the crash can only be captured on the console as once its crashed it can't send network traffic

fcastilloec, 2025-07-16:
I'm seeing the same error as well (`Connection error occurred: [Errno 104] Connection reset by peer`), in addition to the following in Home Assistant:
```
Can't connect to ESPHome API for garage-door @ 192.168.1.90: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.90', port=6053))]: [Errno 111] Connect call failed ('192.168.1.90', 6053) (SocketAPIError)
Can't connect to ESPHome API for garage-door @ 192.168.1.90: Handshake timed out after 30.0s (TimeoutAPIError)
```
The web interface is working intermittently. The status is being reported as `OPEN` then goes `unavailable`, and it's in a very specific pattern, you can see it in this screenshot from Home Assistant, where purple is `OPEN` and blank is `unavailable`. You can see that it stays `OPEN` for around 1 minutes, then goes unavailable for a couple seconds. This happens multiple times until it goes unavailable for 5 minutes, before starting the pattern again.

[image]

I've been using ESPHome 2025.6.3 for over 2 weeks and haven't had any problems until today. I did notice that ESPHome was automatically updated to 2025.7.0, but I reverted the add-on to 2025.6.3, clearing build files and reinstalling, and the issue is still there.

ahembree, 2025-07-16:
I'm also having the same issue described, was also having the issue on 2025.6.3 though and was hoping 2025.7.0 would fix it but nope

Always seems to crash when switching to emulation mode

georgea62, 2025-07-17:
> Can you post a stack trace?
> 
> https://esphome.io/guides/troubleshooting.html

Here:

INFO Starting log output from /dev/cu.usbserial-130 with baud rate 115200
[18:36:30][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:32][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:34][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:36][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:38][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:40][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:42][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:44][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:46][D][time:048]: Synchronized time: 2025-07-16 18:36:46
[18:36:46][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:48][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
[18:36:48][D][ratgdo_secplus1:077]: No wall panel detected. Switching to emulation mode.

[18:36:50]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[18:36:50]Exception (3):
WARNING Exception type: LoadStoreError: Processor internal physical address or data error during load or store
[18:36:50]epc1=0x40225972 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40284a58 depc=0x00000000
WARNING Decoded 0x40225972: esphome::ratgdo::secplus1::Secplus1::wall_panel_emulation(unsigned int)

[18:36:51]>>>stack>>>
WARNING Found stack trace! Trying to decode it

[18:36:51]ctx: cont
[18:36:51]sp: 3ffffdb0 end: 3fffffd0 offset: 0150
[18:36:51]3fffff00:  00000000 0000992e 58d4fdf3 3fff0d24  
[18:36:51]3fffff10:  3fffdad0 00000000 3fff5024 40232371  
WARNING Decoded 0x40232371: esphome::Scheduler::execute_item_(esphome::Scheduler::SchedulerItem*)
[18:36:51]3fffff20:  0000992e 3fff25ac 3fff0774 3fff0d24  
[18:36:51]3fffff30:  3fffdad0 00000000 3fff0774 40233176  
WARNING Decoded 0x40233176: esphome::Scheduler::call()
[18:36:51]3fffff40:  0000000a 00000005 00009929 40246819  
WARNING Decoded 0x40246819: esp_try_delay(unsigned int, unsigned int, unsigned int)
[18:36:51]3fffff50:  00000000 00009929 3ef9db22 00b117b0  
[18:36:51]3fffff60:  00000000 3fff4a14 0000992e 3fff4a14  
[18:36:51]3fffff70:  0000000a 00009929 00000005 3fff0d24  
[18:36:51]3fffff80:  3fffdad0 00000000 3fff0774 4022f13c  
WARNING Decoded 0x4022f13c: esphome::Application::loop()
[18:36:51]3fffff90:  00000001 00000002 00000074 3fff0d24  
[18:36:51]3fffffa0:  3fffdad0 00000000 3fff0cf8 3fff0d24  
[18:36:51]3fffffb0:  3fffdad0 00000000 3fff0cf8 402468c4  
WARNING Decoded 0x402468c4: loop_wrapper() at core_esp8266_main.cpp
[18:36:51]3fffffc0:  feefeffe feefeffe 3fffdab0 40100aad  
WARNING Decoded 0x40100aad: cont_wrapper
[18:36:51]<<<stack<<<

[18:36:51]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[18:36:51] ets Jan  8 2013,rst cause:2, boot mode:(3,1)

[18:36:51]load 0x4010f000, len 3424, room 16 
[18:36:51]tail 0
[18:36:51]chksum 0x2e
[18:36:51]load 0x3fff20b8, len 40, room 8 
[18:36:51]tail 0
[18:36:51]chksum 0x2b
[18:36:51]csum 0x2b
[18:36:51]v0009ef40
[18:36:51]~ld
[18:36:51]\x84\xe3\xecx82'\xec\x93;\x83\xfbg|\xec
                                                 d
                                                  \x84
                                                      d`c\x92x9f|;\x93l\x9co\xe0
                                                                               \xc3'\xe3
        $`\x8c\xe3s\x93d\xc4lx8c
                                \xfe[I][logger:034]: Log initialized
[18:36:51][C][safe_mode:082]: There have been 1 suspected unsuccessful boot attempts
[18:36:51][I][app:049]: Running through setup()
[18:36:51][D][ratgdo:353]: Set opening duration: 13.3s
[18:36:51][D][number:012]: 'Opening duration': Sending state 13.300000
[18:36:51][D][ratgdo:359]: Set closing duration: 12.5s
[18:36:51][D][number:012]: 'Closing duration': Sending state 12.500000
[18:36:51][D][number:012]: 'Client ID': Sending state 7521593.000000
[18:36:51][C][switch.gpio:011]: Running setup for 'Status door'
.......

georgea62, 2025-07-17:
> Thanks [@georgea62](https://github.com/georgea62)
> 
> Would you confirm the exact ESPhome version as well

esphome logs ratgdov25i-47a1b3.yaml
INFO ESPHome 2025.7.0
INFO Reading configuration ratgdov25i-47a1b3.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
INFO Detected timezone 'America/Chicago'
Found multiple options for logging, please choose one:
  [1] /dev/cu.usbserial-130 (USB Serial)
  [2] Over The Air (ratgdov25i-47a1b3.local)
(number): 1
/opt/homebrew/Cellar/esphome/2025.7.0/libexec/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.5 is exactly one major version older than the runtime version 6.31.1 at api.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(
/opt/homebrew/Cellar/esphome/2025.7.0/libexec/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.29.5 is exactly one major version older than the runtime version 6.31.1 at api_options.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(
INFO Starting log output from /dev/cu.usbserial-130 with baud rate 115200
[18:36:30][D][ratgdo_secplus1:070]: Looking for security+ 1.0 wall panel...
......
....

bdraco, 2025-07-17:
ESP8266 only.. Thats the key. Its got to be the PROGMEM change

bdraco, 2025-07-17:
Please flash again and let me know if its fixed. If not, I'll reopen

closed 2025-07-17 by bdraco

ahembree, 2025-07-17:
[image]

Seems to have worked, thank you!

georgea62, 2025-07-17:
It was fast!
Thank you!!!!