ratgdo/esphome-ratgdo #455: Version number and/or build date in ready-to-use firmware file names on the website

armond-avanes, 2025-07-20:
I will be great if some form of version number and/or build date can be added to the file names of the firmware builds on the website: https://ratgdo.github.io/esphome-ratgdo/

When I download a build and before installing it, I have no way to tell whether it's the same version I've downloaded/installed the previous time or it's an updated one.

PaulWieland, 2025-07-20:
Not really practical as the files are generated with static names for the web installer.

closed 2025-07-20 by PaulWieland

iissmart, 2025-07-25:
Would you consider adding the version number, or possibly even just a date it was generated, side-by-side with the file, that the website could show? Or perhaps adding an action to publish a release on https://github.com/ratgdo/esphome-ratgdo/releases once it builds successfully, that we could watch for?

It took me far too long to realize the builds are automatically updated after pushing to main, and I'm a software developer. It would be helpful as a regular user to make this more obvious.