ratgdo/esphome-ratgdo #461: GDO defaults to "open" on reset

dannosaur, 2025-08-09:
Something has changed recently and I'm not sure what. For some reason, the GDO is dropping off the wifi network and resetting itself (potentially, still trying to find the cause of this). But when it comes back online, the door goes to "unknown" initially (which is fine), and then goes to "open" while it's "trying to look for a security+ 1.0 wall panel". It waits for a bit, then gives up, says it's going into emulation mode, sends some bytes, and then the door shows up correctly as closed.

I wouldn't mind, but my Alarmo is also linked to this, so randomly my alarm triggers on the basis that it believes the door has been opened, potentially uncommanded.

Is there a way to make it skip that initial "looking for wall panel..." check that I know fails and go straight to emulation mode, and is there a way to make it report the previous state until it knows for sure what the state of the door is, so that at least it stops my alarm from going off in the middle of the night? I've had to take that entity out of my Alarmo configuration for now. I want to put it back but need to figure out why it randomly reports the door as open when it's not.

kaechele, 2025-08-14:
Disabling Security+ 1.0 auto detection is contained in #304. So that should be fixed when that merges.

dannosaur, 2025-08-14:
@PaulWieland so I actually have the syslog already sending logs out of the ESP so I could catch it whenever it dropped off the network/reset itself. This is the output

[logs-data.csv](https://github.com/user-attachments/files/21781594/logs-data.csv)

(Note: reverse descending order)

The line "Synchronized time" is when it got a network connection back, I don't have anything during the reboot sequence when it wasn't on the network. In HomeAssistant, when it dropped off the network, all the entities became "unavailable" (as expected), but then this happened:

[image]

blulite, 2026-08-18:
Any updates on this ticket?

When new firmware is flashed—or even when the device is simply restarted—the RATGDO reboots and initially defaults to OPEN. It then waits while it searches for the “Looking for security+ 1.0 wall panel…” condition. Once it detects that the door is actually closed, it changes the state to CLOSED.

This is an issue because I have indicators in my home that turn on whenever the garage door is open. As a result, restarting the RATGDO causes those indicators to turn on for roughly 30 seconds before the RATGDO detects the actual door state and turns them back off.

Is there a way to have the RATGDO retain or determine the actual door state before reporting OPEN, or otherwise avoid this temporary false-open state?