ratgdo/esphome-ratgdo #483: Wi-Fi has become buggy

bfish2, 2025-09-17:
I am using a Ratgdo 2.51i that I bought in November of 2024 with a LiftMaster Security 2.0+ garage door.  It has worked well when it's online.

Over the past few months, it has become very slow to come back online after a software update (tens of minutes to an hour or more).   So much so, that I have been hesitant to update it, but I have done so dutifully anyhow.

Today, it didn't come back online at all.  After reading a number of on-line posts, it appeared that I should re-flash it from the https://ratgdo.github.io/esphome-ratgdo/ site.  I did this.  It came right back online while connected to my PC.  But once I connected it to any other power source (USB cable and wall wort), it wouldn't come back online.  I tried several different cables, and several different wall worts.  All to no avail.  It is now sitting connected to my computer and working fine (except of course for not able to control my garage door as it isn't plugged into the wiring harness).

NOTE:  I also tried the steps in the YouTube video about how to confirm the wifi is OK, and it is OK officially.

From the Ratgdo web page:  
Firmware Version | 2025.8.4 Sep 14 2025, 20:43:48

denverpilot, 2025-10-03:
Also seeing this, but not a lot of info yet.  

Access Point(s) are Unifi and switching 2.4 GHz to 20 MHz channel width seemed to help at first just as a sanity check, but only lasted a day or so.  It appears maybe it's rekeying/staying connected to the WiFi that's the problem.

Started sometime in whatever ESPHome build is packaged with Home Assistant and ESPHome Builder 2025.9, currently on 2025.9.3

Going to see if AP reboot brings them back, even temporarily... something about starting the WiFi connection from scratch seems to help, for a while.

denverpilot, 2025-10-03:
> One thing to consider is the use of LED bulbs in the GDO.  Certain brands
> emit RFI at a significant enough level to cause receiver overload on the
> ESP board. The quick test is to remove the bulb and see if the problem
> abates. If you find that is the problem, try a different, name brand, of
> bulb. In the US there are also bulbs specifically for use in GDOs.  There
> maybe in other countries too but I cannot vouch for that.  This, btw, is
> not theory. I experienced this problem awhile back when I replaced a burned
> out bulb in my GDO with an LED bulb.
> […](#)

Good advice, but definitely not my issue. These GDOs have worked fine for nearly a year.  Definitely a software issue in the latest firmware.

lukebarone, 2025-10-05:
Had the same issue. Another thread mentioned to move the antenna from the GDO a bit away from the ratgdo. I used a USB Extension Cable, and it is still up and running (using a ping test). Hopefully it stays working more than a day (it's been a rough week).

bfish2, 2025-10-05:
I just updated yesterday to the latest ESPHome and then updated Ratgdo.  This time it went quickly and flawlessly.

I also have an LED bulb but have had it in there for years.  Nevertheless, next time there's an issue I'll check that.   

Thanks, everyone, for the feedback, and making me feel that there really is an issue and I'm not just crazy!  :)

bfish2, 2025-10-15:
Update:  Updated ESPHome today, and Ratgdo stayed offline.

Rebooted router (AP) and then Ratgdo came back online.

Hoping I don't have to do this forever more!

bdraco, 2025-11-20:
https://esphome.io/changelog/2025.11.0/
https://github.com/esphome/esphome/pull/11805