ratgdo/esphome-ratgdo #485: Ratgdo ESPHOME Log screen constantly scolling -- is this normal??

kevdogg, 2025-09-28:
Ha -- I've had a ratgdo 2.5i sitting in a box for a few years and what possessed my to hook it up this weekend -- I might not even understand.

Anyway - basics
Ratgdo 2.5i, firmware 2025.8.4 Sep 14 2025, 20:43:47 ESPHome Variant, Home Assistant.
I noticed just after flashing firmware and going to ESPHOME page for the device (prior to hooking the garage door up), the log screen constantly scrolling. I also did this after I hooked the physical device to the garage door.

Screen looks like this:

[image]

You can see in the image Multiple Events being recorded per second.
Is this normal? It possibly could be. The device seems to function appropriately without issue by just operating device through ESPHOME web page or through Home Assistant.

restauffer, 2025-09-28:
That is normal. That is how often the GDO is sending out status updates on
those items.

closed 2025-09-28 by PaulWieland

restauffer, 2025-09-28:
> Quite chatty items. Thanks for info. Seems like a lot of traffic being
> generated. Too bad polling or communication interval isn't an adjustable
> parameter.

The "traffic" is coming from the GDO, not the ratgdo,  hence is not
adjustable within ratgdo.